dkogan/mrcal #5: I have a question about cv.undistPoints and cv.remap.

VasilevIvanVladimirovich, 2022-12-15:
Hi @dkogan 
I'm interested in doing an experiment with the camera model: LENSMODEL_SPLINED_STEREOGRAPHIC. I want to see how this model will show itself during the operation of my device.

My application uses the undistPoints function, thanks to which I get the opportunity to create a map of a new undistorted position for each pixel. This allows me to quickly transform at arbitrary points without affecting the entire image.

Using this code, I get an analog of the cv.initUndistortRectifeMap function, and then to get the undistortImage, the cv.remap function is used, which performs the inverse transformation + interpolation.
`m = mrcal.cameramodel('imgPath')
model_pinhole = mrcal.pinhole_model_for_reprojection(m,fit="corners")
mapxy = mral.image_transformation_map(m,model_pinhole )`

I need to get a transformation map for each pixel so that I can quickly know the new location of the undistorted pixel by index.
Blue color - cv.UndistPoint()
Green - mrkal.image_transformation_map()

[image]

How can i do this ?

dkogan, 2022-12-16:
Hi. I don't understand what you're asking. I think you used the code snippet above to visualize
- LENSMODEL_PINHOLE <- LENSMODEL_OPENCV...
- LENSMODEL_PINHOLE <- LENSMODEL_SPLINED_STEREOGRAPHIC_...
Is that what you wanted to do? Are you asking me why the blue and green don't match?

Or you just want a function to map pixel(s) from one model to the other? Assuming extrinsics are identical:

```
p    = mrcal.unproject(q_from, *model_from.intrinsics())
q_to = mrcal.project(p, *model_to.intrinsics())
```
If the two models came from two different calibrations, then you have an implied extrinsics difference to take care of. You need this function: http://mrcal.secretsauce.net/mrcal-python-api-reference.html#-implied_Rt10__from_unprojections

VasilevIvanVladimirovich, 2022-12-19:
Hello @dkogan 
I apologize for the poorly worded question.

When the `mrcal.image_transformation_map()` function is used, I get a reverse map. I can use this map further in the `cv.remap()` function to get an undistorted image. I need to get a undistorted for the point (`cv.remap()` works on the whole image). When I display the map `mrcal.image_transformation_map()`, there is a reverse distortion, different from `cv.undistortPoint()`. 

When I try to translate the coordinate of a point using the map `mrcatl.image_transformation_map()`, I get even more distortion (Example of translating all points):

dkogan, 2022-12-19:
Hi. These functions all map projections from one model to the other. The snippet from the previous message is what you want to do:

```
p    = mrcal.unproject(q_from, *model_from.intrinsics())
q_to = mrcal.project(p, *model_to.intrinsics())
```

This tells you the pixel `q_to` corresponding to the pixel `q_from` in the original model. Set the to/from to whatever models you need. These functions all broadcast, so you can handle many `q_from` at the same time. You can also use the `mrcal-reproject-points` commandline tool if you don't want to write any code.

VasilevIvanVladimirovich, 2023-01-11:
Hi!
Thanks for the reply.
I applied the code above and really got the points I needed.

I have a question, why do I get negative coordinates of points in `cv.undistortPoint()` and in the case of `mrcal-reproject-point` I got all positive coordinates?

[image]

dkogan, 2023-01-11:
If you're giving them the same models and the same data, you should get the same result. Show me the data and code, if you want me to look at it. ALSO, FYI if you're reprojecting images instead of discrete points, the mrcal-reproject-image tool will do all that for you.

VasilevIvanVladimirovich, 2023-01-13:
I was a little confused when I didn't see any negative coordinates. Having tried to do the same, but with a different model, I found the presence of negative coordinates.

Thank you for your help in my question.

closed 2023-01-13 by VasilevIvanVladimirovich