dkogan/mrcal #6: Direct access to state vector uncertainties

rtimpe, 2023-05-08:
Firstly, thanks for all of your work on this toolkit - it's been very helpful!

Looking at the derivation of the projection uncertainty [here](http://mrcal.secretsauce.net/uncertainty.html), I see that `Var(b)` is derived as an intermediate step.  Is it possible to access this information in mrcal?  The projection uncertainty is useful, but we're also interested in propagating the calibration uncertainty to other operations (such as solvePnP).

dkogan, 2023-05-08:
Hello. Usually using =b= directly isn't enough, so I don't expose =Var(b)= directly. All the internal routines that propagate calibration-time uncertainty use this function:

https://github.com/dkogan/mrcal/blob/f866f1b409f72398f84fb6f18635385ce1d1aa21/mrcal/model_analysis.py#L504

This takes a matrix df/db of shape (N,Nstate) and gives you Var(f) of shape (N,N). It's a bit more general than that, actually, but this is the main use case. I suggest you use this function for your uncertainty propagation logic. Do a search to see how other functions use it. This isn't in the "public" interface yet, but I'll add it in the near future.

If you come up with something useful and/or interesting, I'd love to hear about it!

Oh, and in case you weren't aware, mrcal 2.3 was released a few days ago, with much extended documentation. The code and interfaces are largely the same, but you might notice some tweaked things here and there.

rtimpe, 2023-05-08:
> Hello. Usually using =b= directly isn't enough, so I don't expose =Var(b)= directly. All the internal routines that propagate calibration-time uncertainty use this function: [...]

Thanks, I'll take a look at that function.  I'm not entirely sure `Var(b)` is exactly what I want, but good to know about that function.

> 
> Oh, and in case you weren't aware, mrcal 2.3 was released a few days ago, with much extended documentation. The code and interfaces are largely the same, but you might notice some tweaked things here and there.

Yes, the expanded [how to calibrate](http://mrcal.secretsauce.net/how-to-calibrate.html) page is particularly useful.  Thanks!

dkogan, 2023-05-21:
This seems to have been sorted out. I' mclosing the issue. If you still need something, feel free to reopen

closed 2023-05-21 by dkogan