dkogan/mrcal #8: Have you compared the training time of OpenCV‘s parametric model and your splined model in the same dataset?

thunder112358, 2023-08-21:


dkogan, 2023-08-21:
I never did any systematic studies about this, but I have plenty of experience using both kinds of models on the same dataset. The splined model is slower, but not unusably so. In particular, the b-spline representation has local support, so projection at any one spot is affected only by a small subset of parameters. So if you have a parametric model with 10 lens parameters and a splined model with 1000 lens parameters, the splined model computation is NOT 100 times slower. It is far better than that.

You should try it out, since mrcal makes this very simple. And I can imagine that there are plenty of optimization opportunities in the code that would speed it up even more; but so far it hasn't been a problem that needed solving.

dkogan, 2023-10-10:
Closing. Feel free to reopen if there's anything for me to do here.

closed 2023-10-10 by dkogan