dkogan/mrcal #11: mrcal_optimize memory leaks when built with asan

mcm001, 2023-11-19:
I'm using the C api to access mrcal_optimize, using [this code](https://github.com/mcm001/mrcal-java/blob/4a7ebfd3acd2d04323283b85cbdd417444af3345/mrcal_wrapper.cpp#L216). I built libdogleg/mrcal in debug mode locally, using mrcal tag v2.3 and libdogleg HEAD. When I run with asan, I get the following output. I glanced through these and it looks like they're leaks within mrcal, not things I can control from user code? Am I missing code on my end to free objects?

```
mrcal.c(5414): Threw out some outliers. New count = 1/784 (0.1%). Going again

=================================================================
==164458==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 2736 byte(s) in 1 object(s) allocated from:
    #0 0x7f96676b4887 in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f96680052d2 in _dogleg_optimize /home/matt/Documents/GitHub/libdogleg/dogleg.c:1169
    #2 0x7f96680055fa in dogleg_optimize2 /home/matt/Documents/GitHub/libdogleg/dogleg.c:1231
    #3 0x7f9665d157f0 in mrcal_optimize /home/matt/Documents/GitHub/mrcal/mrcal.c:5383
    #4 0x558353e0adc2 in mrcal_main(std::span<mrcal_point3_t, 18446744073709551615ul>, std::span<mrcal_pose_t, 18446744073709551615ul>, cv::Size_<int>, double, cv::Size_<int>) /home/matt/Documents/GitHub/mrcal-java/mrcal_wrapper.cpp:216
    #5 0x558353def06d in homography_test() /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:128
    #6 0x558353def8bd in main /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:161
    #7 0x7f9664c29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

Indirect leak of 366580 byte(s) in 25 object(s) allocated from:
    #0 0x7f96676b4887 in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f9665c24fb5 in cholmod_malloc (/lib/x86_64-linux-gnu/libcholmod.so.3+0x7fb5)

Indirect leak of 8336 byte(s) in 1 object(s) allocated from:
    #0 0x7f96676b4887 in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f9668004ce7 in allocOperatingPoint /home/matt/Documents/GitHub/libdogleg/dogleg.c:1041
    #2 0x7f9668005445 in _dogleg_optimize /home/matt/Documents/GitHub/libdogleg/dogleg.c:1198
    #3 0x7f96680055fa in dogleg_optimize2 /home/matt/Documents/GitHub/libdogleg/dogleg.c:1231
    #4 0x7f9665d157f0 in mrcal_optimize /home/matt/Documents/GitHub/mrcal/mrcal.c:5383
    #5 0x558353e0adc2 in mrcal_main(std::span<mrcal_point3_t, 18446744073709551615ul>, std::span<mrcal_pose_t, 18446744073709551615ul>, cv::Size_<int>, double, cv::Size_<int>) /home/matt/Documents/GitHub/mrcal-java/mrcal_wrapper.cpp:216
    #6 0x558353def06d in homography_test() /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:128
    #7 0x558353def8bd in main /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:161
    #8 0x7f9664c29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

Indirect leak of 8336 byte(s) in 1 object(s) allocated from:
    #0 0x7f96676b4887 in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f9668004ce7 in allocOperatingPoint /home/matt/Documents/GitHub/libdogleg/dogleg.c:1041
    #2 0x7f9668005423 in _dogleg_optimize /home/matt/Documents/GitHub/libdogleg/dogleg.c:1197
    #3 0x7f96680055fa in dogleg_optimize2 /home/matt/Documents/GitHub/libdogleg/dogleg.c:1231
    #4 0x7f9665d157f0 in mrcal_optimize /home/matt/Documents/GitHub/mrcal/mrcal.c:5383
    #5 0x558353e0adc2 in mrcal_main(std::span<mrcal_point3_t, 18446744073709551615ul>, std::span<mrcal_pose_t, 18446744073709551615ul>, cv::Size_<int>, double, cv::Size_<int>) /home/matt/Documents/GitHub/mrcal-java/mrcal_wrapper.cpp:216
    #6 0x558353def06d in homography_test() /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:128
    #7 0x558353def8bd in main /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:161
    #8 0x7f9664c29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

Indirect leak of 104 byte(s) in 1 object(s) allocated from:
    #0 0x7f96676b4887 in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f9668004c4f in allocOperatingPoint /home/matt/Documents/GitHub/libdogleg/dogleg.c:1028
    #2 0x7f9668005445 in _dogleg_optimize /home/matt/Documents/GitHub/libdogleg/dogleg.c:1198
    #3 0x7f96680055fa in dogleg_optimize2 /home/matt/Documents/GitHub/libdogleg/dogleg.c:1231
    #4 0x7f9665d157f0 in mrcal_optimize /home/matt/Documents/GitHub/mrcal/mrcal.c:5383
    #5 0x558353e0adc2 in mrcal_main(std::span<mrcal_point3_t, 18446744073709551615ul>, std::span<mrcal_pose_t, 18446744073709551615ul>, cv::Size_<int>, double, cv::Size_<int>) /home/matt/Documents/GitHub/mrcal-java/mrcal_wrapper.cpp:216
    #6 0x558353def06d in homography_test() /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:128
    #7 0x558353def8bd in main /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:161
    #8 0x7f9664c29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

Indirect leak of 104 byte(s) in 1 object(s) allocated from:
    #0 0x7f96676b4887 in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f9668004c4f in allocOperatingPoint /home/matt/Documents/GitHub/libdogleg/dogleg.c:1028
    #2 0x7f9668005423 in _dogleg_optimize /home/matt/Documents/GitHub/libdogleg/dogleg.c:1197
    #3 0x7f96680055fa in dogleg_optimize2 /home/matt/Documents/GitHub/libdogleg/dogleg.c:1231
    #4 0x7f9665d157f0 in mrcal_optimize /home/matt/Documents/GitHub/mrcal/mrcal.c:5383
    #5 0x558353e0adc2 in mrcal_main(std::span<mrcal_point3_t, 18446744073709551615ul>, std::span<mrcal_pose_t, 18446744073709551615ul>, cv::Size_<int>, double, cv::Size_<int>) /home/matt/Documents/GitHub/mrcal-java/mrcal_wrapper.cpp:216
    #6 0x558353def06d in homography_test() /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:128
    #7 0x558353def8bd in main /home/matt/Documents/GitHub/mrcal-java/mrcal_test.cpp:161
    #8 0x7f9664c29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

SUMMARY: AddressSanitizer: 386196 byte(s) leaked in 30 allocation(s).
```

mcm001, 2023-11-19:
Update: after I applied this diff, the asan leak warnings are no longer present. Looks like the context was being reallocated after discarding outliers and the initial one was never freed?

```
index 447ed577..2e319b66 100644
--- a/mrcal.c
+++ b/mrcal.c
@@ -5380,6 +5380,9 @@ mrcal_optimize( // out
         double outliernessScale = -1.0;
         do
         {
+            // Make sure we don't reallocate the context by accident?
+            if(solver_context != NULL) dogleg_freeContext(&solver_context);
+            
             norm2_error = dogleg_optimize2(packed_state,
                                            Nstate, ctx.Nmeasurements, ctx.N_j_nonzero,
                                            (dogleg_callback_t*)&optimizer_callback, &ctx,
```

mcm001, 2023-11-19:
I can also reproduce it with the mrcal/libdogleg from apt (not my locally compiled one) by just setting `do_apply_outlier_rejection` to false in my code, if that's helpful

dkogan, 2023-11-19:
Thank you very much for finding this. It's a clear bug; here's the relevant documentation for `dogleg_optimize2()`: https://github.com/dkogan/libdogleg/blob/3d45754325f9ef1c9c6e8851d09a1cfb749572c4/dogleg.h#L184

I pushed the update to git. I had no plans to do any point releases for 2.3 with bug fixes (such as this one). SHOULD I make point releases? Would it make your life easier if there was a mrcal 2.3.1 that fixes things like this?

closed 2023-11-19 by dkogan

mcm001, 2023-11-19:
Thanks for the quick reply! I don't think this is a blocker for me yet -- can tolerate a leak as long as I don't call calibrate too many times. Do you think that mrcal will have another full release at some point before January? If not, it could be nice to get this fix back ported by then.

dkogan, 2023-11-19:
Matt ***@***.***> writes:

> Do you think that mrcal will have another full release at some point
> before January?

No way. 3.0 will have some big improvements, and they're nowhere near
done.

> If not, it could be nice to get this fix back ported by then.

OK. I'll make a point release at some point in the next few weeks. Will
let you know.