dkogan/mrcal #15: Install after building from source

handysome6, 2024-03-19:
After building, the generated libray and executables are located in the original folder. In order to execute like installed from apt source, I have to add the folder to the system path. Also the python bind of the mrcal library is also not installed to the given PYTHONPATH. 
I have to add the folder to the python's sys.path.

So I am wondering if we can add more building option that can support `make install`? Or is there any way that can install the built python bind into some environment?

dkogan, 2024-03-19:
Hi. There is a "make install" rule you can use. Try something like

```
mkdir /tmp/dir
DESTDIR=/tmp/dir make install
```

Does that do what you want? If not, then I'd like a bit more clarity on what you're hoping for.

handysome6, 2024-03-20:
Thanks for your reply. This command `DESTDIR=/tmp/dir make install` is useful to me. I see this command will install `bin`, `inculde`, `lib` and `share` in the `usr` folder under the given `DESTDIR` path. 

If I change the `DESTDIR` to the root, I can install all mrcal files into the corresponding locations as the `apt-get` install method. However, I didn't find a way to undo this installation. Ways that I tried:
```
1. make uninstall                                                    # no such rule
2. cat install_manifest.txt | xargs echo rm | sh        # no install_manifest file
```

Currently I am using the python `mrcal` module. So I just added the `mrcal` folder to the `sys.path` in python file, which works file. Yet installation remains a problem if undoing is not possible.

dkogan, 2024-03-20:
You want packages. The mrcal install page describes where to get prebuilt ones. Are those insufficient for you? If so, why? You can build your own packages, if you want to, but tell me what you want, first.

handysome6, 2024-03-25:
Sorry for the late reply. 

There are bugs in the pre-built apt source version of mrcal when calibrating stereo camera pairs. So I choose to build from source under tag number v2.4. The build was successful and work normally. 

The only question is that I want to use `import mrcal` in my python scripts. And I couldn't find a way to install the built mrcal python package to a python library path. Currently I am modifying the `sys.path` variable before importing this module. This works with no problem. I am only wondering if there is a way to avoid such dirty method. Thanks!

dkogan, 2024-03-26:
Hello. If there are bugs in the provided packages, please report those here. There should not be any bugs.

Changing the `sys.path` (or the `PYTHONPATH`) shouldn't be enough: you should also need to set the `LD_LIBRARY_PATH`. In any case, if you're not installing the build, setting the paths is what you should do. If you want to install, you should use packages. And if you want to install stuff yourself, you're making a mess of your machine that you should try to avoid. But it's your machine, so you can certainly do that: copy the build to any path that's already in your `sys.path`. Or something like `DESTDIR=/ make install` will probably do it. But that makes a mess and I don't recommend it.

handysome6, 2024-03-27:
Thanks for your answer. I encountered bugs when using the mrcal package installed from original apt sources in Ubuntu 22.04 Jammy, the version of which is 2.1-5build1. After I add your source `deb [trusted=yes] http://mrcal.secretsauce.net/packages/DISTRO/public/ DISTRO main` to `/etc/apt/sources.list` and the mrcal version turns to 2.4.1 and the problem disappered. 

Thanks for your suggestions, I will take that. Adding the folder to `PYTHONPATH` did worked for me, but I don't need to build from source myself now, since the apt source version works great. 

Your help is greatly appreciated. Thanks for sharing such useful tools!

dkogan, 2024-03-27:
Andy Liu ***@***.***> writes:

> Thanks for your answer. I encountered bugs when using the mrcal
> package installed from original apt sources in Ubuntu 22.04 Jammy, the
> version of which is 2.1-5build1. After I add your source deb [...]

I still don't know what the problem was. Is everybody seeing a broken
package?

> Thanks for your suggestions, I will take that. Adding the folder to
> PYTHONPATH did worked for me, but I don't need to build from source
> myself now, since the apt source version works great.
>
> Your help is greatly appreciated. Thanks for sharing such useful tools!

OK. Glad you're finding it useful!

closed 2024-03-27 by dkogan