dkogan/mrcal #16: Differencing extrinsic-stability using stereo systems

handysome6, 2024-03-27:
Thank your for your patience about the package issue. I have another problem regarding the extrinsic stability as should in [https://mrcal.secretsauce.net/differencing.html](url)

I am trying to replicate your extrinsic-stability result in the linked page. This is my setup:

1. 20MP cameras: 1.1inch sensor with C-mount lenses, resolution: 5472*3648, FOV 63
2. baseline: roughly 1.0m long aluminum square tube, with cameras fixed on left and right
3. calibration senerio: 8*11 chessboard, box size 60mm, first calib intrinsics, then calib extrinsics with intrinsic solve skipped
4. feature extration using Matlab(only for extracting corners) to get corners.vnl
5. Image number for calibration each camera's intrinsic is ~45
6. Number of image pairs for calibrating extrinsic is ~35
```
# calib intrinsic command
mrcal-calibrate-cameras          \
  --corners-cache [LR]-corners.vnl    \
  --corners-cache-has-weights    \
  --lensmodel  LENSMODEL_SPLINED_STEREOGRAPHIC_order=3_Nx=33_Ny=22_fov_x_deg=60 \
  --focal 5000                          \
  --object-spacing 0.0600       \
  --object-width-n 8               \
  --object-height-n 11            \
  '*.jpg'

# calib extrinsic command
mrcal-calibrate-cameras          \
  --corners-cache stereoCorners.vnl    \
  --corners-cache-has-weights    \
  --seed splined_A.cameramodel,splined_D.cameramodel  \
  --object-spacing 0.0600        \
  --object-width-n 8             \
  --object-height-n 11           \
  --skip-intrinsics-solve        \
  --pairs     'A_*.jpg' 'D_*.jpg'
```
The pixel error in calibration stays low. However, after carrying out several individual expriment of continous calibrating, the extrinsic-stability output is not good as the given example in the linked website. 

The expiriment is carried out in a sequence:
```
calib1a->calib1b->external strike->
calib2a->calib2b->external strike->
calib3...
```

Now I am still stuck at reporducing a clean result without external impact as you showed in the website, i.e., comparing the extrinsic stability of calib1a/1b, calib2a/2b and extra. A typical ouput is like this:
```
translation: 1.91mm in the direction [-0.2   0.08 -0.98]
rotation:    0.014deg around the axis [0.29 0.61 0.74]
```

This result is apparently not resonable as I didn't touch the stereo system between the tested two calibrations. The translation difference range is about [0.5, 3] in millimeters. If the extrinsic changed about 2mm, the baseline would have visible deformation, which I didn't spot any. 

Could you given more details about your camera setup> Such as about resolution, baseline, image number and extra. And if you have other suggestions, I am more than eager to accept.

dkogan, 2024-03-31:
OK. I have questions.

You say you ran two commands. Is that correct? How do you get the two
sets of intrinsics? You first command, as written, computes one model
only.

Why are you splitting any of this at all? Why not do one solve with all
your data, computing the intrinsics and extrinsics together? They affect
each other and you will get biases if you split them in this way.

The extrinsics difference is reported comparing the coordinate systems
of each camera. The origin of these coordinate systems isn't fixed
inside the mechanical camera unit: it cam move around a bit, so a 1mm
shift could maybe not be unreasonable.

As noted in the documentation, the extrinsics-stability tooling isn't
thoroughly tested yet. If you redo your experiment without the split
solve, and send me your data, I can look at it. Eventually. I suspect
that your solve has some minor issues that are giving you those
extrinsics shifts, but it would take too much typing to list out the
various potential problems. The tooling should run those checks for you,
but today we're not quite there yet.

handysome6, 2024-04-08:
Thanks for the suggestion. I've successfully reduced the extrinsic difference by combining the vnl files into one and calibrate the stereo system in a single command. 

The reduced translation difference is around 0.1 or so. The reduced rotaion difference is around 0.01 degree. Although the difference is still larger than the demoed one. I think this one is resonable. I will carry out more expirements later. Thank you!

dkogan, 2024-04-09:
OK. I'm going to close this. Reopen if you want to re-examine your data

closed 2024-04-09 by dkogan