dkogan/mrcal #20: Question about Projection Uncertainty calculation and valid intrinsics region

handysome6, 2024-04-25:
Reading the offical document of mrcal here. [https://mrcal.secretsauce.net/uncertainty.html](url) I think high of the idea that visualizing the sampling error by calculating the projection uncertainty. 

However, there is one limitation of the projection uncertainty, as you put it here:

> Currently the uncertainty estimates can be computed only from a vanilla calibration problem: **a set of stationary cameras** observing a moving calibration object. Other formulations can be used to compute the lens parameters as well (**structure-from-motion** while also computing the lens models for instance), but at this time the uncertainty computations cannot handle those cases. It can be done, but the current method needs to be extended to do so.

I want to move (pitch and yaw) the camera rig while moving the checkerboard as well in the 'dancing' process. Under such situation, is it still possible to use the projection uncertainty function to show sampling error? I think that cameras are not 'stationary' in this process, yet I can't say that I am using structure-from-motion method for calibrating cameras. 

The cameras are firmly fixed onto a rigid body, the relative position didn't change during the image taking procedure. I hope that the checkerboard chould be viewed as 'dancing' around relative to the cameras. So that I can use projection uncertainty in this case.

dkogan, 2024-04-26:
You will not be able to apply the uncertainty method in mrcal 2.4. But not only that, you'll have great trouble calibrating at all. Do you have this implemented already? How? Does it work?

handysome6, 2024-04-26:
No, I haven't implement it. But I did try doing something similar in a hand-held device, which is also a stereo vision system. I calibrate it using the same checkerboard. Two person is involved in this process, one hold the camera rig, while the other carrying the board. The camera man will rotate the camera rig to make sure checker pattern appears in every corner of the image, while the checkerboard is also rotated to get different angle of views of the pattern. 

This is done before I try using mrcal. The calibration runs well in opencv and Matlab's camera calibration process. 

Actually I think there isn't too much different to a stationary setup of cameras group. In both cases we all have similar transformation of checkerboard positions relative to the camera setup.

dkogan, 2024-04-27:
Oh; the cameras all move as a rigid set. You said that, but I didn't notice it. In that case, it will all work for the reason you state: because you can always think of the chessboard geometry relative to the camera rig, ignoring the fact that the cameras are moving in some global frame. Yeah, it'll all work ok.

handysome6, 2024-05-02:
Thanks for your confirmation! I will try this in the following calibration process.

closed 2024-05-02 by handysome6