dkogan/mrcal #22: cannot build in anaconda enviroment

Baehn, 2024-08-10:
I encountered a linker error when trying to build the library under an anaconda enviroment (wihtout anaconda it works). The issue seems to be with the libpython3.11.a static library, which the linker cannot find.  
Below is the error with sensitive paths obscured:                                                                                                                             
  
```bash                                                                                                                                                                           
 gcc -pthread ... -L/home/XXXXX/miniconda3/envs/mrcal/lib libpython3.11.a ... -o mrcal/_mrcal.cpython-311-x86_64-linux-gnu.so                                                
 /home/XXXXX/miniconda3/envs/mrcal/compiler_compat/ld: cannot find libpython3.11.a: No such file or directory                                                                
 collect2: error: ld returned 1 exit status                                                                                                                                  
 make: *** [Makefile:136: mrcal/_mrcal.cpython-311-x86_64-linux-gnu.so] Error 1                                                                                              
```                                                                                                                                                                             
The -L option is used correctly to specify the library search path, but the libpython3.11.a part seems to be incorrectly placed in the command. It should be linked with     
-lpython3.11 if the shared library is to be used, or the path to the static library should be corrected if it exists.

I have installed libpython3.11.a file in `/home/XXXXX/miniconda3/envs/mrcal/lib` and changing the command manually to the correct format works.

dkogan, 2024-08-10:
Hi. Thanks for the report. I'm not entirely clear on what's happening for you. Can I get the FULL link command that is giving you the error? You abbreviated it above, and the full thing would clarify stuff for me.

dkogan, 2024-08-10:
In addition to the above, can you please run `python3` and then `import sysconfig` and then tell me what these say:

```
print(sysconfig.get_config_var('BLDSHARED'))
print(sysconfig.get_config_var('BLDLIBRARY'))
print(sysconfig.get_config_var('LDLIBRARY'))
print(sysconfig.get_config_var('LDFLAGS'))
```

I might have missed one, but these hopefully will be enough.

Thanks

Baehn, 2024-08-11:
here is the complete command:

 ```bash                                                                                                                                                                     
gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat -shared -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib libpython3.11.a -Wl,-rpath,'$ORIGIN'/.. -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/repros/mrcal -L/home/XXXXX/miniconda3/envs/mrcal/lib -lpython3.11 mrcal-pywrap.o -lmrcal -lsuitesparseconfig -o mrcal/_mrcal.cpython-311-x86_64-linux-gnu.so

```

here is the output you requested:

```bash
gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat -shared  -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib
libpython3.11.a
libpython3.11.a
-Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib

```

dkogan, 2024-08-13:
Thanks for the reply. Getting closer. The build system (mrbuild) uses the options python tells it to use (the sysconfig things above). But I'm probably not using the exact right options in the exact right order.

Your command has `-lpython3.11`, but that doesn't appear in the set of sysconfig options you reported. Is this something you added manually? If not, can I get ALL the sysconfig options? `sysconfig.get_config_vars()`. That would tell me where it's coming from.

Assuming you DID add it manually, let's try a fix. In `mrcal/Makefile` can you change `$(PY_MRBUILD_LINKER) ... -o $@` to `$(PY_BLDSHARED) ... -o $@ $(PY_BLDLIBRARY)`? There should be two instances, and it should fix it. Can you confirm?

This isn't THE fix, but a confirmation would tell me what to do.

Thanks

Baehn, 2024-08-13:
sorry, this was my fault, I changed it manually to `-lpython3.11`, here is the command again:
```bash
gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat -shared -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib libpython3.11.a -Wl,-rpath,'$ORIGIN'/.. -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/repros/mrcal  mrcal-pywrap.o -lmrcal -lsuitesparseconfig -o mrcal/_mrcal.cpython-311-x86_64-linux-gnu.so
```

I did the changes to the Makefile you suggested, which results in:
```bash
gcc -pthread -B /home/XXXX/miniconda3/envs/mrcal/compiler_compat -shared -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXX/miniconda3/envs/mrcal/lib -L/home/XXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXX/miniconda3/envs/mrcal/lib -L/home/XXXX/miniconda3/envs/mrcal/lib -Wl,-rpath,'$ORIGIN'/.. -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXX/miniconda3/envs/mrcal/lib -L/home/XXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXX/miniconda3/envs/mrcal/lib -L/home/XXXX/miniconda3/envs/mrcal/lib -L/home/XXXX/repros/mrcal  mrcal-pywrap.o -lmrcal -o mrcal/_mrcal.cpython-311-x86_64-linux-gnu.so libpython3.11.a 
/home/XXXX/miniconda3/envs/mrcal/compiler_compat/ld: cannot find libpython3.11.a: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:136: mrcal/_mrcal.cpython-311-x86_64-linux-gnu.so] Error 1
```

changing it to 
```bash
gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat -shared -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath,'$ORIGIN'/.. -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/repros/mrcal  mrcal-pywrap.o -lmrcal -o mrcal/_mrcal.cpython-311-x86_64-linux-gnu.so /home/XXXXX/miniconda3/envs/mrcal/lib/libpython3.11.a -fno-lto
```
made this pass, note that i also added `-fno-lto`

dkogan, 2024-08-13:
Still doesn't work? Sigh. OK. I'm going to ask for more info, and then, hopefully I can fix it without asking you for anything more.

- I'd like ALL the sysconfig options: `sysconfig.get_config_vars()`
- And I'd like to get the "right" build flags by seeing what setuptools does. You can do that by building a hello-world extension. Clone `git@github.com:starnight/python-c-extension.git`. Then `cd python-c-extension/00-HelloWorld` and `python3 setup.py build`

That build command will report what it does. I'd like a full copy of the build command it reports. I tried to read the setuptools code to figure out what it does, but it's too complex to pull the logic from there.

Thanks

Baehn, 2024-08-15:
here is the output from the helloworld extension:
```bash
running build
running build_ext
building 'helloworld' extension
creating build
creating build/temp.linux-x86_64-cpython-311
gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/XXXXX/miniconda3/envs/mrcal/include -fPIC -O2 -isystem /home/XXXXX/miniconda3/envs/mrcal/include -fPIC -I/home/XXXXX/miniconda3/envs/mrcal/include/python3.11 -c bind.c -o build/temp.linux-x86_64-cpython-311/bind.o
gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/XXXXX/miniconda3/envs/mrcal/include -fPIC -O2 -isystem /home/XXXXX/miniconda3/envs/mrcal/include -fPIC -I/home/XXXXX/miniconda3/envs/mrcal/include/python3.11 -c libmypy.c -o build/temp.linux-x86_64-cpython-311/libmypy.o
creating build/lib.linux-x86_64-cpython-311
gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat -shared -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib -Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib -L/home/XXXXX/miniconda3/envs/mrcal/lib build/temp.linux-x86_64-cpython-311/bind.o build/temp.linux-x86_64-cpython-311/libmypy.o -o build/lib.linux-x86_64-cpython-311/helloworld.cpython-311-x86_64-linux-gnu.so
```

here is the full sysconfig output:
```bash
{'ABIFLAGS': '',
 'AC_APPLE_UNIVERSAL_BUILD': 0,
 'AIX_BUILDDATE': 0,
 'AIX_GENUINE_CPLUSPLUS': 0,
 'ALIGNOF_LONG': 8,
 'ALIGNOF_SIZE_T': 8,
 'ALT_SOABI': 0,
 'ANDROID_API_LEVEL': 0,
 'AR': 'ar',
 'ARFLAGS': 'rcs',
 'BASECFLAGS': '',
 'BASECPPFLAGS': '-IObjects -IInclude -IPython',
 'BASEMODLIBS': '',
 'BINDIR': '/home/XXXXX/miniconda3/envs/mrcal/bin',
 'BINLIBDEST': '/home/XXXXX/miniconda3/envs/mrcal/lib/python3.11',
 'BLDLIBRARY': 'libpython3.11.a',
 'BLDSHARED': 'gcc -pthread -B '
              '/home/XXXXX/miniconda3/envs/mrcal/compiler_compat -shared  '
              '-Wl,--allow-shlib-undefined '
              '-Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib '
              '-Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib '
              '-L/home/XXXXX/miniconda3/envs/mrcal/lib '
              '-Wl,--allow-shlib-undefined '
              '-Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib '
              '-Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib '
              '-L/home/XXXXX/miniconda3/envs/mrcal/lib',
 'BOOTSTRAP_HEADERS': '\\',
 'BUILDEXE': '',
 'BUILDPYTHON': 'python',
 'BUILD_GNU_TYPE': 'x86_64-conda_cos6-linux-gnu',
 'BYTESTR_DEPS': '\\',
 'CC': 'gcc -pthread -B /home/XXXXX/miniconda3/envs/mrcal/compiler_compat',
 'CCSHARED': '-fPIC',
 'CFLAGS': '-DNDEBUG -fwrapv -O2 -Wall   -fPIC   -O2 -isystem '
           '/home/XXXXX/miniconda3/envs/mrcal/include -fPIC -O2   -isystem '
           '/home/XXXXX/miniconda3/envs/mrcal/include ',
 'CFLAGSFORSHARED': '',
 'CFLAGS_ALIASING': '',
 'CONFIGFILES': 'configure configure.ac acconfig.h pyconfig.h.in '
                'Makefile.pre.in',
 'CONFIGURE_CFLAGS': '-fPIC -O2 -isystem '
                     '/home/XXXXX/miniconda3/envs/mrcal/include ',
 'CONFIGURE_CFLAGS_NODIST': '-fno-semantic-interposition  -g -std=c11 '
                            '-Werror=implicit-function-declaration '
                            '-fvisibility=hidden',
 'CONFIGURE_CPPFLAGS': '-DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -isystem '
                       '/home/XXXXX/miniconda3/envs/mrcal/include '
                       '-I/home/XXXXX/miniconda3/envs/mrcal/include',
 'CONFIGURE_LDFLAGS': '-Wl,--allow-shlib-undefined '
                      '-Wl,-rpath,/home/XXXXX/miniconda3/envs/mrcal/lib '
                      '-Wl,-rpath-link,/home/XXXXX/miniconda3/envs/mrcal/lib '
                      '-L/home/XXXXX/miniconda3/envs/mrcal/lib',
 'CONFIGURE_LDFLAGS_NODIST': '-fno-semantic-interposition  -g',
 'CONFIGURE_LDFLAGS_NOLTO': '-fno-lto',
 'CONFIG_ARGS': "'--prefix=/home/XXXXX/miniconda3/envs/mrcal''--build=x86_64-conda_cos6-linux-gnu''--host=x86_64-conda_cos6-linux-gnu''--enable-ipv6''--with-ensurepip=no''--with-tzpath=/home/XXXXX/miniconda3/envs/mrcal/share/zoneinfo:/home/XXXXX/miniconda3/envs/mrcal/share/tzinfo''--with-computed-gotos''--with-system-ffi''--with-system-expat''--enable-loadable-sqlite-extensions''--with-tcltk-includes=-I/home/XXXXX/miniconda3/envs/mrcal/include''--with-tcltk-libs=-L/home/XXXXX/miniconda3/envs/mrcal/lib "
[... 1128 lines cut ...]
 'WITH_LIBINTL': 0,
 'WITH_NEXT_FRAMEWORK': 0,
 'WITH_PYMALLOC': 1,
 'WITH_VALGRIND': 0,
 'X87_DOUBLE_ROUNDING': 0,
 'XMLLIBSUBDIRS': 'xml xml/dom xml/etree xml/parsers xml/sax',
 'abiflags': '',
 'abs_builddir': '/home/conda/feedstock_root/build_artifacts/python-split_1713550854359/work/build-static',
 'abs_srcdir': '/home/conda/feedstock_root/build_artifacts/python-split_1713550854359/work',
 'base': '/home/XXXXX/miniconda3/envs/mrcal',
 'datarootdir': '/home/XXXXX/miniconda3/envs/mrcal/share',
 'exec_prefix': '/home/XXXXX/miniconda3/envs/mrcal',
 'installed_base': '/home/XXXXX/miniconda3/envs/mrcal',
 'installed_platbase': '/home/XXXXX/miniconda3/envs/mrcal',
 'platbase': '/home/XXXXX/miniconda3/envs/mrcal',
 'platlibdir': 'lib',
 'prefix': '/home/XXXXX/miniconda3/envs/mrcal',
 'projectbase': '/home/XXXXX/miniconda3/envs/mrcal/bin',
 'py_version': '3.11.9',
 'py_version_nodot': '311',
 'py_version_nodot_plat': '',
 'py_version_short': '3.11',
 'srcdir': '/home/XXXXX/miniconda3/envs/mrcal/lib/python3.11/config-3.11-x86_64-linux-gnu',
 'userbase': '/home/XXXXX/.local'}
```

dkogan, 2024-08-23:
Thank you very much for including all of that! It looks like the "right" way to build it is to not link in `libpython` at all. That's not strictly correct: `_mrcal.cpython-312-x86_64-linux-gnu.so` uses symbols in `libpython`, so it should link against it. But it will only ever be used from within a `python` process, and those symbols will be available anyway.

So let's try this. If you confirm that this works, I'll commit the change to mrbuild. In `mrcal/Makefile` can you change `$(PY_MRBUILD_LINKER) ... -o $@` to `$(PY_BLDSHARED) ... -o $@`? So same as before, but without adding `$(PY_BLDLIBRARY)` to the end. I claim this will allow you to build and run mrcal stuff.

Thanks for helping track this down.

dkogan, 2024-09-04:
I think the fix committed into mrbuild 1.12 works, so I'm closing this. Please reopen, if it has issues. Thanks

closed 2024-09-04 by dkogan