dkogan/mrcal #27: Usage of different detectors

VeeranjaneyuluToka, 2024-10-02:
Hi, Looks to me this is a very nice tool for camera cailbration!

Wondering if there is any samples on how to use detector other than mrgingham. I had a look into doc, looks like there is some process to create .vnl file but still not clear how to create mrcal compatable .vnl file by running other detectors for example opencv findchessboardcorners function. 

Thanks!

dkogan, 2024-10-02:
Are the docs unclear? The page is here: https://mrcal.secretsauce.net/recipes.html#non-mrgingham-detector It notes what the assumptions and expectations are, and includes an example data file. What else would you like?

VeeranjaneyuluToka, 2024-10-03:
Thanks again for quick reply! i was wondering if there is any possible flow as below
-> Run opencv detector to get corners
-> Run vnlog to create .vnl file
So wondering if there is any api from vnlog to create .vnl files.

VeeranjaneyuluToka, 2024-10-03:
I just creaetd a txt file with required format and changed extention. It works, thanks!