dkogan/mrcal #28: calibrate cameras fails

VeeranjaneyuluToka, 2024-10-02:
I have 8x7 checkerboard target and each cell size is 10x10 cm.

I have used mrgingham detector by passing gridn as 6 and it was detecting corners as shown below
[image]
[image]

but when i run the the below command to calibrate.
mrcal-calibrate-cameras --corners-cache corners.vnl --lensmodel LENSMODEL_OPENCV8 --focal 1029 --object-spacing 0.1 --object-width-n 6 '*.JPG'

It fails with the below error
"Error extracting or reading chessboard corners: Found too few (0; need at least 1) images containing a calibration pattern in camera 0; glob '*/*.JPG'"

When i had a look into implementation, it fails to compute chessboard corners, wondering how come it worked when i use detector alone and fails when i try to calibrate and does it work with this kind of target at all? Thanks!

dkogan, 2024-10-02:
Hello. First of all, mrgingham assumes square boards, and you can see that in the detections you're showing here: one row of corners is ignored. It won't be a consistent row every time. This board cannot be used with mrgingham.

The question you're actually asking is separate. In the error message you pasted it says `glob '/.JPG'`. Is that actually the error message? If so, the glob has been changed from `*.JPG` somewhere. In any case, I'm happy to debug if you give me the `corners.vnl` and let me know which version of mrcal you're using. But again, this won't work: this board must be square.

VeeranjaneyuluToka, 2024-10-03:
FYI, I changed the command as follows
mrcal-calibrate-cameras --corners-cache corners_cv.vnl --lensmodel LENSMODEL_OPENCV8 --focal 1029 --object-spacing 0.1 --object-width-n 6 --object-height-n 7 '*.jpg'

It works, the change is JPG to jpg. Thanks!

closed 2024-10-03 by VeeranjaneyuluToka