dkogan/mrcal #31: Error when running propagation uncertainty using test-triangulation-uncertainty.py

posadad1, 2024-11-06:
Hi, I am currently trying to evaluate the uncertainty propagation a stereo camera setup I have with a baseline of 1.5m and a range of 600m. I am trying to use `test/test-projection-uncertainty.py` for this purpose with the following parameters:
`python3 repos/mrcal/test/test-triangulation-uncertainty.py --baseline 1.5 --observed-point -2 0 600 --q-calibration-stdev 0.2 --q-observation-stdev 0.2`. I had to deactivate the local mrcal (Line 215-218 ) because it was not finding the python3 mrcal library.

I tried the default sample:

`test/test-triangulation-uncertainty.py  \
  --do-sample                           \
  --cache write                         \
  --observed-point -2 0 10              \
  --fixed cam0                          \
  --Nsamples 200                        \
  --Ncameras 2                          \
  --q-observation-stdev-correlation 0.5 \
  --q-calibration-stdev 0.2             \
  --q-observation-stdev 0.2             \
  --make-documentation-plots '' `
 
After I did this and ran it, it kept giving me the following error and  I have not been able to find the root case even if I try the default examples.

[image]

dkogan, 2024-11-06:
Hi. Before I say anything more: which version of mrcal are you running, specifically?

posadad1, 2024-11-06:
I did not compile it but installed it using the command line so the version is:
``` 
libmrcal-dev is already the newest version (2.1-5build1).
mrcal is already the newest version (2.1-5build1).
python3-mrcal is already the newest version (2.1-5build1).
```

posadad1, 2024-11-06:
I am almost done but have been trying to solve some issues with the dependencies. I installed everything per the documentation for building `python3-gl-image-display` but when I go ahead and do make I get the following error
```
g++ -Wall -Wextra -I/usr/local/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_THREAD_SAFE -D_REENTRANT -Wno-missing-field-initializers -Wno-unused-parameter -MMD -MP -g -fno-omit-frame-pointer -DMRBUILD_VERSION='"0.1"' -fPIC -std=c++0x -O3 -c -o Fl_Gl_Image_Widget.o Fl_Gl_Image_Widget.cc
< image.vertex.glsl sed 's/.*/"&\\n"/g' > image.vertex.glsl.h.tmp && mv image.vertex.glsl.h.tmp image.vertex.glsl.h
< image.geometry.glsl sed 's/.*/"&\\n"/g' > image.geometry.glsl.h.tmp && mv image.geometry.glsl.h.tmp image.geometry.glsl.h
< image.fragment.glsl sed 's/.*/"&\\n"/g' > image.fragment.glsl.h.tmp && mv image.fragment.glsl.h.tmp image.fragment.glsl.h
< line.vertex.glsl sed 's/.*/"&\\n"/g' > line.vertex.glsl.h.tmp && mv line.vertex.glsl.h.tmp line.vertex.glsl.h
< line.geometry.glsl sed 's/.*/"&\\n"/g' > line.geometry.glsl.h.tmp && mv line.geometry.glsl.h.tmp line.geometry.glsl.h
< line.fragment.glsl sed 's/.*/"&\\n"/g' > line.fragment.glsl.h.tmp && mv line.fragment.glsl.h.tmp line.fragment.glsl.h
cc -Wall -Wextra --std=gnu99 -Wno-missing-field-initializers -Wno-unused-parameter -MMD -MP -g -fno-omit-frame-pointer -DMRBUILD_VERSION='"0.1"' -fPIC -O3 -c -o GL_image_display.o GL_image_display.c
g++ -shared  -fPIC -Wl,-soname,libGL_image_display_fltk.so.0  -Wl,-rpath='$ORIGIN'/ -Wl,--default-symver -shared -fPIC -Wl,-soname,libGL_image_display.so.0 GL_image_display.o  -lGLU -lGL -lepoxy -lglut -lfreeimage -lm -pthread -lfltk_gl -lfltk -lX11 -o libGL_image_display.so.0.1
ln -fs libGL_image_display.so.0.1 libGL_image_display.so
g++ -shared  -fPIC -Wl,-soname,libGL_image_display_fltk.so.0  -Wl,-rpath='$ORIGIN'/ Fl_Gl_Image_Widget.o libGL_image_display.so -lGLU -lGL -lepoxy -lglut -lfreeimage -lm -pthread -lfltk_gl -lfltk -lX11 -o libGL_image_display_fltk.so.0.1
ln -fs libGL_image_display_fltk.so.0.1 libGL_image_display_fltk.so
ln -fs libGL_image_display_fltk.so.0.1 libGL_image_display_fltk.so.0
swig \
  -w302 -w312 -w325 -w362 -w389 -w401 -w473 -w509 \
  -I/usr/include/ \
  -DFL_EXPORT="" \
  -DPYTHON \
  -DPYTHON3 \
  -python \
  -c++ \
  -keyword \
  -shadow \
  -fastdispatch \
  -outdir . \
  -o Fl_Gl_Image_Widget_pywrap.cc \
  Fl_Gl_Image_Widget.i
g++ -Wall -Wextra -I/usr/local/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_THREAD_SAFE -D_REENTRANT -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -fPIC -I/usr/include/python3.10 -Wno-missing-field-initializers -Wno-unused-parameter -MMD -MP -g -fno-omit-frame-pointer -DMRBUILD_VERSION='"0.1"' -std=c++0x -O3 -c -o Fl_Gl_Image_Widget_pywrap.o Fl_Gl_Image_Widget_pywrap.cc
x86_64-linux-gnu-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -g -fwrapv -O2  -lpython3.10  -Wl,-rpath,'$ORIGIN'/ -Wl,-Bsymbolic-functions -g -fwrapv -O2  -L/home/dposada/repos/mrcal_stuff/GL_image_display -Wl,-rpath=$ORIGIN/ Fl_Gl_Image_Widget_pywrap.o libGL_image_display_fltk.so.0.1 -o _Fl_Gl_Image_Widget.cpython-310-x86_64-linux-gnu.so
ln -fs libGL_image_display.so.0.1 libGL_image_display.so.0
cc -Wall -Wextra --std=gnu99 -Wno-missing-field-initializers -Wno-unused-parameter -MMD -MP -g -fno-omit-frame-pointer -DMRBUILD_VERSION='"0.1"' -O3 -c -o GL_image_display-test-glut.o GL_image_display-test-glut.c
g++ -Wl,-rpath,'$ORIGIN'/  GL_image_display-test-glut.o libGL_image_display.so -lGLU -lGL -lepoxy -lglut -lfreeimage -lm -pthread -o GL_image_display-test-glut
g++ -Wall -Wextra -I/usr/local/include -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_THREAD_SAFE -D_REENTRANT -Wno-missing-field-initializers -Wno-unused-parameter -MMD -MP -g -fno-omit-frame-pointer -DMRBUILD_VERSION='"0.1"' -std=c++0x -O3 -c -o GL_image_display-test-fltk.o GL_image_display-test-fltk.cc
g++ -Wl,-rpath,'$ORIGIN'/  GL_image_display-test-fltk.o libGL_image_display_fltk.so.0.1 libGL_image_display.so -lGLU -lGL -lepoxy -lglut -lfreeimage -lm -pthread -lfltk_gl -lfltk -lX11 -o GL_image_display-test-fltk
/usr/bin/ld: GL_image_display-test-fltk.o: in function `Fl_Gl_Window::pixel_w()':
/usr/local/include/FL/Fl_Gl_Window.H:229: undefined reference to `Fl_Gl_Window::pixels_per_unit()'
/usr/bin/ld: GL_image_display-test-fltk.o: in function `Fl_Gl_Window::pixel_h()':
/usr/local/include/FL/Fl_Gl_Window.H:239: undefined reference to `Fl_Gl_Window::pixels_per_unit()'
/usr/bin/ld: GL_image_display-test-fltk.o: in function `Fl_Gl_Window::pixel_w()':
/usr/local/include/FL/Fl_Gl_Window.H:229: undefined reference to `Fl_Gl_Window::pixels_per_unit()'
/usr/bin/ld: GL_image_display-test-fltk.o: in function `Fl_Gl_Window::pixel_h()':
/usr/local/include/FL/Fl_Gl_Window.H:239: undefined reference to `Fl_Gl_Window::pixels_per_unit()'
/usr/bin/ld: GL_image_display-test-fltk.o: in function `Fl_Gl_Window::pixel_w()':
/usr/local/include/FL/Fl_Gl_Window.H:229: undefined reference to `Fl_Gl_Window::pixels_per_unit()'
/usr/bin/ld: GL_image_display-test-fltk.o:/usr/local/include/FL/Fl_Gl_Window.H:239: more undefined references to `Fl_Gl_Window::pixels_per_unit()' follow
[... cut ...]
/usr/bin/ld: GL_image_display-test-fltk.o:(.data.rel.ro._ZTV26Fl_Gl_Image_Widget_Derived[_ZTV26Fl_Gl_Image_Widget_Derived]+0x88): undefined reference to `Fl_Group::on_remove(int)'
/usr/bin/ld: GL_image_display-test-fltk.o:(.data.rel.ro._ZTV26Fl_Gl_Image_Widget_Derived[_ZTV26Fl_Gl_Image_Widget_Derived]+0x90): undefined reference to `Fl_Group::delete_child(int)'
collect2: error: ld returned 1 exit status
make: *** [mrbuild/Makefile.common.footer:226: GL_image_display-test-fltk] Error 1

```

I did install `libfltk1.3-dev` and pyFLTK through Pypi. Any idea what could be causing this? I would love to have the the vis tools for the mrcal-stereo.

posadad1, 2024-11-06:
I tried building it without it just to check if it would build: This is how bot my paths look like 
```
$PATH: other_stuff:/home/dposada/repos/mrcal_stuff/:/home/dposada/repos/mrcal_stuff/mrgingham/:/home/dposada/repos/mrcal_stuff/re2c-3.1/:/home/dposada/repos/mrcal_stuff/libelas/

$PYTHONPATH: /home/dposada/repos/mrcal_stuff/numpysane/:/home/dposada/repos/mrcal_stuff/gnuplotlib/
```

However I got into a new error when building mrcal 
```
< CHOLMOD_factorization.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > CHOLMOD_factorization.docstring.h
< CHOLMOD_factorization_rcond.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > CHOLMOD_factorization_rcond.docstring.h
< CHOLMOD_factorization_solve_xt_JtJ_bt.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > CHOLMOD_factorization_solve_xt_JtJ_bt.docstring.h
< corresponding_icam_extrinsics.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > corresponding_icam_extrinsics.docstring.h
< drt_ref_refperturbed__dbpacked.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > drt_ref_refperturbed__dbpacked.docstring.h
< knots_for_splined_models.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > knots_for_splined_models.docstring.h
< lensmodel_metadata_and_config.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > lensmodel_metadata_and_config.docstring.h
< lensmodel_num_params.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > lensmodel_num_params.docstring.h
< load_image.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > load_image.docstring.h
< measurement_index_boards.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > measurement_index_boards.docstring.h
< measurement_index_points.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > measurement_index_points.docstring.h
< measurement_index_regularization.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > measurement_index_regularization.docstring.h
< num_intrinsics_optimization_params.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_intrinsics_optimization_params.docstring.h
< num_measurements_boards.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_measurements_boards.docstring.h
< num_measurements.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_measurements.docstring.h
< num_measurements_points.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_measurements_points.docstring.h
< num_measurements_regularization.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_measurements_regularization.docstring.h
< num_states_calobject_warp.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_states_calobject_warp.docstring.h
< num_states.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_states.docstring.h
< num_states_extrinsics.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_states_extrinsics.docstring.h
< num_states_frames.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_states_frames.docstring.h
< num_states_intrinsics.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_states_intrinsics.docstring.h
< num_states_points.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > num_states_points.docstring.h
< optimize.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > optimize.docstring.h
< optimizer_callback.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > optimizer_callback.docstring.h
< pack_state.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > pack_state.docstring.h
< _rectification_maps.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > _rectification_maps.docstring.h
< _rectified_resolution.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > _rectified_resolution.docstring.h
< _rectified_system.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > _rectified_system.docstring.h
< save_image.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > save_image.docstring.h
< state_index_calobject_warp.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > state_index_calobject_warp.docstring.h
< state_index_extrinsics.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > state_index_extrinsics.docstring.h
< state_index_frames.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > state_index_frames.docstring.h
< state_index_intrinsics.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > state_index_intrinsics.docstring.h
< state_index_points.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > state_index_points.docstring.h
< supported_lensmodels.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > supported_lensmodels.docstring.h
< traverse_sensor_links.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > traverse_sensor_links.docstring.h
< unpack_state.docstring sed 's/\\/\\\\/g; s/"/\\"/g; s/^/"/; s/$/\\n"/;' > unpack_state.docstring.h
cc -Wall -Wextra --std=gnu99 -I/usr/include/suitesparse -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -fPIC -I/usr/include/python3.10 -Wno-cast-function-type -Wno-int-conversion -Wno-missing-field-initializers -Wno-unused-variable -Wno-unused-parameter -Wno-missing-braces -MMD -MP -g -fno-omit-frame-pointer -DMRBUILD_VERSION='"2023-07-26--triangulated-features-merged-423-gecf219e8"' -O3 -c -o mrcal-pywrap.o mrcal-pywrap.c
[... cut ...]
mrcal-pywrap.c:3227:2: warning: #warning "triangulated-solve: barf if we have any triangulated points" [-Wcpp]
 3227 | #warning "triangulated-solve: barf if we have any triangulated points"
      |  ^~~~~~~
mrcal-pywrap.c: At top level:
mrcal-pywrap.c:4317:10: fatal error: measurement_index_points_triangulated.docstring.h: No such file or directory
 4317 | #include "measurement_index_points_triangulated.docstring.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [mrbuild/Makefile.common.footer:153: mrcal-pywrap.o] Error 1

```

dkogan, 2024-11-06:
Any particular reason you're building anything? The APT server has all
the dependencies.

> I did install libfltk1.3-dev and pyFLTK through Pypi

As a general rule, never install anything from pypi, or any other source
that isn't the distro or distro-adjacent. You're using untested version
combinations of things if you do that.

posadad1, 2024-11-06:
You are absolutely right, I just realized I did not do the Pypi version but the recommended from them which is the APT  `
    sudo apt install python3-fltk
` but this is the one giving me this errors.

dkogan, 2024-11-06:
Sorry, I don't follow. You now installed everything from apt, and it's all good now? If not, what doesn't work? You CAN build everything yourself if you want, but that's more work, and you shouldn't need to do that.

dkogan, 2024-11-06:
Daniel Posada ***@***.***> writes:

> However I got into a new error when building mrcal
>
> mrcal-pywrap.c:4317:10: fatal error: measurement_index_points_triangulated.docstring.h: No such file or directory
>  4317 | #include "measurement_index_points_triangulated.docstring.h"
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.

You're trying to build the bleeding edge code from git. This isn't
released yet, and isn't expected to work or build or anything. Unless
you really need some of the half-done new stuff, stick with the latest
release: 2.4.1. There are packages. And a git tag, if you REALLY like
building things.

posadad1, 2024-11-06:
I installed all the dependencies I could from APT source :
- libdogleg-dev
- python3-yaml
- python3-fltk
- libfreeimage-dev

I was able to build successfully the following:
- mrbuild
- vnlog
- re2c-3.1
- libelas
- mrgingham

I cloned the python ones:
- python3-numpysane
- python3-gnuplotlib

I have errors building:
- python3-gl-image-display
- mrcal

The [first error post](https://github.com/dkogan/mrcal/issues/31#issuecomment-2460603466) is the `python3-gl-image-display` error, the [second error post](https://github.com/dkogan/mrcal/issues/31#issuecomment-2460678794) is the `mrcal` error.

posadad1, 2024-11-06:
> Daniel Posada ***@***.***> writes:
> However I got into a new error when building mrcal mrcal-pywrap.c:4317:10: fatal error: measurement_index_points_triangulated.docstring.h: No such file or [...]

The simpler the better hehe is 2.4.1 stored on the 2.4 branch?

posadad1, 2024-11-06:
> Can you not simply "apt install mrcal" and be 100% done? Why are you building anything?

Ah! I see what you mean, the mrcal version available through apt was the 2.1.5 to [begin with](https://github.com/dkogan/mrcal/issues/31#issuecomment-2460477485)

dkogan, 2024-11-06:
Ah. No. 2.1.5 is what's in your distro (which is several years old). My apt repo provides the newer versions of the dependencies and of the thing itself. Type "apt install mrcal". You're done.

posadad1, 2024-11-06:
> Ah. No. 2.1.5 is what's in your distro (which is several years old). My apt repo provides the newer versions of the dependencies and of the thing itself. Type "apt install mrcal". You're done.

I see, I am in Ubuntu22.04 so the distro most available package is 2.1.5. I will try an enable universe and see if I can install the 2.1.4 https://manpages.ubuntu.com/manpages/noble/man1/mrcal-calibrate-cameras.1.html

dkogan, 2024-11-06:
No. You're done already. You added my repo, as directed on the mrcal install page? Yes? Then you're done. "apt install mrcal"

posadad1, 2024-11-06:
> No. You're done already. You added my repo, as directed on the mrcal install page? Yes? Then you're done. "apt install mrcal"

Thanks for being so patient I see, when I looked into `deb [trusted=yes] http://mrcal.secretsauce.net/packages/DISTRO/public/ DISTRO main` my brain immediately said only for debian and completely skipped it!

 I just installed 2.4.1 but hit the same roadblock when running
```
python3 repos/mrcal/test/test-triangulation-uncertainty.py --baseline 1.5 --observed-point -2 0 600 --q-calibration-stdev 0.2 --q-observation-stdev 0.2
```

the output it gives me is:
```
Traceback (most recent call last):
  File "/home/dposada/repos/mrcal_stuff/mrcal/test/test-triangulation-uncertainty.py", line 347, in <module>
    calibration_baseline(args.model,
  File "/home/dposada/repos/mrcal_stuff/mrcal/test/test_calibration_helpers.py", line 264, in calibration_baseline
    synthesize(z             = range_to_boards,
  File "/home/dposada/repos/mrcal_stuff/mrcal/test/test_calibration_helpers.py", line 239, in synthesize
    mrcal.synthesize_board_observations(models_true_refcam0,
TypeError: synthesize_board_observations() got an unexpected keyword argument 'pcamera_nominal_ref'

```

dkogan, 2024-11-06:
Heh. You installed mrcal 2.4.1, but you're running a test from the bleeding edge mrcal. `git reset --hard v2.4.1`

posadad1, 2024-11-06:
> Heh. You installed mrcal 2.4.1, but you're running a test from the bleeding edge mrcal. `git reset --hard v2.4.1`

Ah ha! Dummy me, I changed branch to release-2.4 but did not reset! Thank you so much! This solved the issue. all tests passed <3

posadad1, 2024-11-06:
and I am back! When running the example:

```
python3 test-triangulation-uncertainty.py  \
  --do-sample                           \
  --cache write                         \
  --observed-point -2 0 10              \
  --fixed cam0                          \
  --Nsamples 200                        \
  --Ncameras 2                          \
  --q-observation-stdev-correlation 0.5 \
  --q-calibration-stdev 0.2             \
  --q-observation-stdev 0.2             \
  --make-documentation-plots ''
```
This is the output:
```
test-triangulation-uncertainty.py:416: OK: Noiseless triangulation should be perfect
test-triangulation-uncertainty.py:431: OK: Re-optimized triangulation should be close to the reference. This checks the regularization bias
test-triangulation-uncertainty.py:518: OK: Gradient check: dp_triangulated_dbstate[intrinsics0]
test-triangulation-uncertainty.py:524: OK: Gradient check: dp_triangulated_dbstate[intrinsics1]
test-triangulation-uncertainty.py:538: OK: Gradient check: dp_triangulated_dbstate[extrinsics1]
test-triangulation-uncertainty.py:549: OK: Gradient check: dp_triangulated_drtrf
test-triangulation-uncertainty.py:573: OK: Gradient check: dp_triangulated_dq: cross-point terms are 0
test-triangulation-uncertainty.py:577: OK: Gradient check: dp_triangulated_dq
test-triangulation-uncertainty.py:609: OK: triangulate(no noise) returns the right point
test-triangulation-uncertainty.py:613: OK: triangulate(cal noise) returns the right point
test-triangulation-uncertainty.py:617: OK: triangulate(obs noise) returns the right point
test-triangulation-uncertainty.py:621: OK: triangulate(both noise) returns the right point
test-triangulation-uncertainty.py:625: OK: triangulate(cal noise) returns a consistent Var_p_calibration
test-triangulation-uncertainty.py:629: OK: triangulate(obs noise) returns a consistent Var_p_observation
test-triangulation-uncertainty.py:633: OK: Var_p_joint.shape matches Var_p_calibration.shape
test-triangulation-uncertainty.py:638: OK: Var(joint) should be Var(cal-time-noise) + Var(obs-time-noise)
test-triangulation-uncertainty.py:638: OK: Var(joint) should be Var(cal-time-noise) + Var(obs-time-noise)
Sampling 20/200
Sampling 40/200
Sampling 60/200
Sampling 80/200
Sampling 100/200
Sampling 120/200
Sampling 140/200
Sampling 160/200
Sampling 180/200
Sampling 200/200
Wrote cache to /tmp/test-triangulation-uncertainty--cam0-opencv4-50-200-2-0-1-0-0.2-0.2-0.5.pickle
test-triangulation-uncertainty.py:782: OK: Triangulated position matches sampled mean
test-triangulation-uncertainty.py:791: OK: Var(dq) is symmetric for triangulated point variance for point 0
test-triangulation-uncertainty.py:791: OK: Var(dq) worst[0] eigenvalue match for triangulated point variance for point 0
test-triangulation-uncertainty.py:791: OK: Var(dq) worst[1] eigenvalue match for triangulated point variance for point 0
test-triangulation-uncertainty.py:791: OK: Var(dq) worst[2] eigenvalue match for triangulated point variance for point 0
test-triangulation-uncertainty.py:791: OK: Var(dq) eigenvectors match for triangulated point variance for point 0
test-triangulation-uncertainty.py:791: OK: Var(dq) is symmetric for triangulated point variance for point 1
test-triangulation-uncertainty.py:791: OK: Var(dq) worst[0] eigenvalue match for triangulated point variance for point 1
test-triangulation-uncertainty.py:791: OK: Var(dq) worst[1] eigenvalue match for triangulated point variance for point 1
test-triangulation-uncertainty.py:791: OK: Var(dq) worst[2] eigenvalue match for triangulated point variance for point 1
test-triangulation-uncertainty.py:791: OK: Var(dq) eigenvectors match for triangulated point variance for point 1
Traceback (most recent call last):
  File "/home/dposada/repos/mrcal_stuff/mrcal/test/test-triangulation-uncertainty.py", line 820, in <module>
    [ mrcal.utils._plot_args_points_and_covariance_ellipse(p_triangulated_sampled0[:,ipt,(0,2)],
  File "/home/dposada/repos/mrcal_stuff/mrcal/test/test-triangulation-uncertainty.py", line 820, in <listcomp>
    [ mrcal.utils._plot_args_points_and_covariance_ellipse(p_triangulated_sampled0[:,ipt,(0,2)],
  File "/usr/lib/python3/dist-packages/mrcal/utils.py", line 1252, in _plot_args_points_and_covariance_ellipse
    return ( _plot_arg_covariance_ellipse(q_mean,Var, what),
  File "/usr/lib/python3/dist-packages/mrcal/utils.py", line 1234, in _plot_arg_covariance_ellipse
    l,v   = _sorted_eig(Var)
NameError: name '_sorted_eig' is not defined. Did you mean: 'sorted_eig'?

```

posadad1, 2024-11-06:
Checking the source the definition is `def sort_eig()` , but when called it was with `_sort_eig()`, could this be a typo?

dkogan, 2024-11-07:
You probably found a bug; thanks. This has been long fixed in the bleeding-edge mrcal. I guess I can release a mrcal 2.4.2 with this bug fix. But nobody complained yet, so maybe it's not a huge issue. Are things working for you after your little patch?

posadad1, 2024-11-07:
Yes, so far with everything yes. I can see the uncertainty propagation now. Feel free to close the issue if that is an acceptable solution :) Thank you for your time

dkogan, 2024-11-07:
OK. I'm going to close. If anybody else says something, I'll release 2.4.2, but I kinda doubt that will happen.

Thanks.

closed 2024-11-07 by dkogan