dkogan/mrcal #32: Incomplete chessboard detection

cassianokc, 2024-11-19:
Hi, first of all thank you very much for the tool and the in depth documentation / calibration troubleshooting guide.
Performing the calibration dance with two big baseline cameras can be challenging because all chessboard corners should be detected. We are using Charuco boards.

Although the documentation states clearly that all chessboard corners should be detected, the [calibration.py comments](https://github.com/dkogan/mrcal/blob/0665efc32ff3eba27ea7956e03a4f759d2708314/mrcal/calibration.py#L130) mentions that one could give weight 0 in order to support incomplete detections. Would this work?

dkogan, 2024-11-21:
The `mrgingham` detector requires the whole chessboard to be visible in order for the detection to succeed. But `mrcal` does not require this detector to be used. You can utilize any other detector, and feed the (possibly-incomplete) observations to mrcal. See https://mrcal.secretsauce.net/recipes.html#non-mrgingham-detector

Let me know if you have questions. If not, please close the report. Thanks

dkogan, 2026-09-06:
Closing; reopen if needed

closed 2026-09-06 by dkogan