dkogan/mrcal #34: Rectification Map Rotates Image 180 degrees

VivaSuresh, 2024-12-04:
After a stereo calibration procedure, I used mrcal.rectified_system and mrcal.rectification_maps to get two rectification maps for the left and right cameras, and then stored that output as a csv. For some reason, after applying these on the camera streams, it rotated the images 180 degrees. The stereo map was still very high quality, so the calibration worked. I have used the same exact code for stereo calibration before and this is the first time this has ever happened. 

For context on the command executed for calibration: 

mrcal-calibrate-cameras --corners-cache ./corners_test.vnl \ 
--lensmodel LENSMODEL_PINHOLE --focal 1000 --object-spacing 0.03 \ 
-- object-width-n 17 'data/stereo_quarter/frame*_left.png' 'data/stereo_quarter/frame*_right.png'

I was curious as to why this would happen?

dkogan, 2024-12-05:
The first rectified camera is always to the LEFT of the second rectified camera. You gave it your RIGHT camera and then your LEFT camera, so the rectification turned the images upside-down to satisfy its condition. If you don't want this, give it the LEFT camera first.

VivaSuresh, 2024-12-05:
Hi! Thank you very much for the response. 

Turns out that the left and right cameras were swapped in the housing, and that is why this happened. It is amazing that your software was able to correct for that. 

Thank you again!

dkogan, 2024-12-05:
Glad you figured it out. I'm closing the issue.

closed 2024-12-05 by dkogan