dkogan/mrcal #35: Issues with lens with big radial distortion parameter

tholzmann, 2025-03-11:
Hello,

First of all I want to thank you for your great calibration tool and your very insightful documentation! 

Coming to my issue: I want to estimate the intrinsics of a camera with 75mm focal length. The camera does not have severe lens distortion, but due to the fact that the focal length is quite large, the second radial distortion parameter (I'm using OPENCV4 model) is assumed to be in between -15-20.

However, when calibrating with regularization turned on, the radial distortion parameters are not growing that much. They stay at -1 or -2 max.

When I turn off regularization, I get the expected result. However, i get an error about estimating valid-intrinsics region:

>mrcal-calibrate-cameras --corners-cache /mnt/d/data/calib_experiments/2025-03-11_test1/boofcv_detections/corners.vnl --outdir /mnt/d/data/calib_experiments/2025-03-11_test1/noregularization --skip-regularization --lensmodel LENSMODEL_OPENCV4 --focal 22990 --object-width-n 8 --object-height-n 11 --object-spacing 0.06 '/mnt/d/data/calib_experiments/2025-03-11_test1/images_png/*.png'
>## initial solve: geometry only
>## RMS error: 0.53
>
>## initial solve: geometry and LENSMODEL_STEREOGRAPHIC core only
>=================== optimizing everything except board warp from seeded intrinsics
>mrcal.c(5564): Threw out some outliers. New count = 105/57376 (0.2%). Going again
>mrcal.c(5564): Threw out some outliers. New count = 132/57376 (0.2%). Going again
>## final, full optimization
>mrcal.c(5564): Threw out some outliers. New count = 170/57376 (0.3%). Going again
>## RMS error: 0.25
>RMS reprojection error: 0.2 pixels
>Worst residual (by measurement): 1.3 pixels
>Noutliers: 170 out of 28688 total points: 0.6% of the data
>calobject_warp = [-0.00033422 -0.00131892]
>
>WARNING: Couldn't compute valid-intrinsics region for camera 0. Will continue without. Error: unsupported operand type(s) for +: 'NoneType' and >'int'
>Wrote /mnt/d/data/calib_experiments/2025-03-11_test1/noregularization/camera-0.cameramodel

A valid calibration is written to disk, just the valid-intrinsics region is missing in the file. The estimated intrinsics are:

> 'intrinsics': [ 21735.40488, 21721.88716, 2120.131743, 1069.908169, 0.01998406516, -16.56381503, -0.0003251621748, -0.0008952591931,],

However, when I want to show the projection uncertainty, I get this error:

>mrcal-show-projection-uncertainty noregularization/camera-0.cameramodel --cbmax 1 --unset key
>Traceback (most recent call last):
>  File "/usr/bin/mrcal-show-projection-uncertainty", line 205, in <module>
>    plot = mrcal.show_projection_uncertainty(model,
>  File "/usr/lib/python3/dist-packages/mrcal/visualization.py", line 1508, in show_projection_uncertainty
>    err = mrcal.projection_uncertainty(pcam * (distance if distance is not None else 1.0),
>  File "/usr/lib/python3/dist-packages/mrcal/model_analysis.py", line 1132, in projection_uncertainty
>    return _propagate_calibration_uncertainty(what,
>  File "/usr/lib/python3/dist-packages/mrcal/model_analysis.py", line 674, in _propagate_calibration_uncertainty
>    if imeas_regularization + Nmeasurements_regularization != Nmeasurements_all:
>TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'

I also tried to manually add the valid intrinsics region field, but this didn't change anything.

Is it possible that the large radial distortion parameter causes issues? If so, how could I fix this?

Thank you,
Thomas

dkogan, 2025-03-11:
Hi. Thanks for the reply. Would it be possible for me to get your input
data so that I can try to reproduce this on my end?

tholzmann, 2025-03-11:
Sure. Is it enough to send you the detected corners? There are quite some images (approx 350).

tholzmann, 2025-03-11:
Sure, I'll send you the data soon via Email.

tholzmann, 2025-03-12:
Update: I have these problems on WSL using ubuntu 22.04. When I run the same commands on a "real" linux with Ubuntu 24.04, I don't have these issues.

dkogan, 2025-03-13:
Thanks for the data. You found a bug! I never disable the regularization, so never noticed this. The attached patch will fix it. If/when I release a 2.4.2 I'll include this fix, but I'd rather not make a new release just for this. Can you live with the extra patch?

I don't understand the OS difference you're seeing. In Ubuntu you don't see the None+int errors? I only tested on Linux, and saw what you described.

Now about regularization. It looks like for this data the solve is a bit over-regularized. Let me know if skipping the regularization works for you. If not, I can tell you how to adjust the weitghts. I'll adjust the weights for the next release. Would it be ok if I used this data in the test suite to check the fix?

Thanks.

[fix-no-regularization-solve.patch.txt](https://github.com/user-attachments/files/19222386/fix-no-regularization-solve.patch.txt)

tholzmann, 2025-03-14:
Yes, exactly. I don't have the None+int error in regularization in the "real" LInux (not WSL). However, I just observed that I have installed an older version of mrcal there (2.1, the other one with the issue was 2.4.1). I guess this was the reason.

So thanks for the patch! I'll try to apply it on the latest version and verify if it also works for me.

Regarding the data: There might be some data in the images which should not be made publically available (e.g. number plates). I'll check the data, and will otherwise provide you a different but very similar dataset where I have no concerns.

tholzmann, 2025-03-14:
Ok, then it's for sure not a problem. Feel free to use it!

tholzmann, 2025-03-14:
And yes, please replace the filenames with something generic. Thx!