dkogan/mrcal #36: AttributeError when calibrating with a non-mrgingham corner detector

DanielSimsTOA, 2025-03-13:
Greetings! 

First, I want to thank you again for making such a useful tool. It is fundamental to my work. 

I'm using a non-mrgingham corner detector. I'm able to run a python script to make a corners.vnl file. When I run "mrcal-calibrate-cameras --corners-cache corners.vnl '*.jpg'" I get the following error: 

`mrcal-calibrate-cameras --corners-cache corners.vnl '*.jpg'
Traceback (most recent call last):
  File "/usr/bin/mrcal-calibrate-cameras", line 256, in <module>
    file = sys.stdderr)
AttributeError: module 'sys' has no attribute 'stdderr'. Did you mean: 'stderr'?`

Am I doing something wrong?

dkogan, 2025-03-13:
Hi. Yes. You're using a very old version of mrcal. I'm guessing you're on 2.1? The latest stable release is 2.4.1. See the install page on how to get it, if your distro doesn't provide it: https://mrcal.secretsauce.net/install.html

DanielSimsTOA, 2025-03-13:
Thanks for the fast reply! Let me install 2.4.1.

DanielSimsTOA, 2025-03-13:
For some reason, I can't upgrade mrcal. 

"$ sudo apt install mrcal
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
mrcal is already the newest version (2.1-5build1).
0 upgraded, 0 newly installed, 0 to remove and 111 not upgraded."

I'm running in a supported Ubuntu environment. 

"lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy"

Am I doing something wrong? Apologies for the low level question.

dkogan, 2025-03-13:
You should read about to how use the package manager; it will be extremely useful. Probably do an `apt update` before the `apt install`

closed 2025-03-14 by DanielSimsTOA