dkogan/mrcal #37: Large RMS reproject error when using a non-mrgingham corner detector

DanielSimsTOA, 2025-03-14:
Greetings! 

Thank you once again for making such a useful tool and for helping with my previous problem. I have some code that goes from an opencv checkerboard detector to a detected_corners.vnl file. I'm able to feed this into mrcal, but I'm getting a reprojection error of 232.53 pixels. This makes, if I understand correctly, the calibration invalid. 

Here is my detected corners file. 

`#   filename      x    y
4.jpg 529.099976 222.289993
4.jpg 609.710022 223.339996
4.jpg 690.229980 224.410004
4.jpg 770.039978 225.449997
4.jpg 849.549988 226.229996
4.jpg 928.750000 227.139999
4.jpg 1008.030029 227.649994
4.jpg 1087.479980 228.250000
4.jpg 1166.260010 228.630005
4.jpg 1245.349976 228.789993
4.jpg 1324.859985 229.089996
4.jpg 1404.229980 229.380005
4.jpg 1483.189941 229.979996
4.jpg 1561.310059 230.729996
4.jpg 1638.969971 231.820007
4.jpg 531.510010 303.420013
4.jpg 611.549988 304.390015
4.jpg 691.479980 305.339996
4.jpg 770.859985 306.250000
4.jpg 849.869995 306.690002
4.jpg 928.609985 307.390015
4.jpg 1007.309998 307.980011
4.jpg 1086.380005 308.369995
4.jpg 1164.510010 308.970001
4.jpg 1242.859985 309.239990
4.jpg 1321.729980 309.619995
4.jpg 1400.199951 310.170013
4.jpg 1478.619995 310.390015
4.jpg 1556.810059 310.859985
4.jpg 1634.520020 311.750000
4.jpg 533.559998 383.309998
4.jpg 613.150024 384.170013
4.jpg 692.030029 384.799988
4.jpg 771.109985 385.390015
4.jpg 849.900024 385.899994
4.jpg 928.320007 386.369995
4.jpg 1006.640015 387.200012
4.jpg 1084.949951 387.529999
4.jpg 1162.609985 388.149994
4.jpg 1239.839966 388.679993
4.jpg 1317.729980 388.959991
4.jpg 1395.310059 389.380005
4.jpg 1473.910034 389.459991
4.jpg 1552.199951 389.769989
4.jpg 1629.560059 390.510010
4.jpg 535.349976 462.450012
4.jpg 614.380005 463.109985
4.jpg 693.169983 463.559998
4.jpg 771.780029 464.190002
4.jpg 850.369995 464.489990
4.jpg 928.109985 464.829987
4.jpg 1005.659973 465.339996
4.jpg 1083.609985 465.630005
4.jpg 1160.020020 466.260010
4.jpg 1236.790039 466.459991
4.jpg 1314.410034 466.779999
4.jpg 1392.099976 467.049988
4.jpg 1470.569946 467.480011
4.jpg 1548.609985 467.760010
4.jpg 1625.310059 468.480011
4.jpg 537.159973 540.559998
4.jpg 616.169983 540.849976
4.jpg 694.580017 541.250000
[... 371 lines cut ...]
6.jpg 1674.459961 728.619995
6.jpg 943.460022 634.719971
6.jpg 975.609985 642.409973
6.jpg 1010.960022 650.340027
6.jpg 1048.530029 658.510010
6.jpg 1088.609985 667.299988
6.jpg 1130.699951 676.520020
6.jpg 1175.869995 686.489990
6.jpg 1224.069946 696.969971
6.jpg 1274.829956 708.340027
6.jpg 1329.030029 720.549988
6.jpg 1387.709961 734.070007
6.jpg 1450.310059 748.309998
6.jpg 1517.680054 763.520020
6.jpg 1589.290039 779.880005
6.jpg 1666.489990 797.479980
`

Attached are the photos. 

[image]
[image]
[image]

What am I doing wrong?

dkogan, 2025-03-14:
Can you please post the exact command you're feeding this data to?

DanielSimsTOA, 2025-03-14:
After further review, I set focal to the focal length in meters, not pixels. This caused the error. Thank you so much for your patience and help.