dkogan/mrcal #39: Multi-camera calibration

zumpchke, 2025-04-23:
I am trying to achieve multi-camera calibration. There are N=7 cameras arranged in a ring in a 360 fashion with equal spacing. Clearly, all cameras are unable to see the board at a given time. How best to achieve calibration for all intrinsics/extrinsics?

Intrinsics could be done separately.

For extrinsics, would this approach work?

- Pairwise calibration of cam0 and cam1
   - Take X number of poses with the board at different tilt angles (Seen only by cam0 and cam1)
  
- Pairwise calibration of cam1 and cam2
   - Take X number of poses with the board at different tilt angles  (Seen only by cam1 and cam2)

.. repeat for camY, camY+1

Does mrcal contain the smarts to get all extrinsics for this setup?

dkogan, 2025-04-23:
Is there enough overlap between adjacent cameras to get chessboard observations in there? If so, it will all work fine. You want the whole set of cameras to be connected transitively; every camera doesn't need to share observations with every other camera

zumpchke, 2025-04-23:
So this scheme would work in corners.vnl (since I'm generating my own corners). Say I take 3 poses per camera pair

```
frame000-camera000.png
frame000-camera001.png

frame001-camera000.png
frame001-camera001.png

frame002-camera000.png
frame002-camera001.png

^^^^ Now I would shift the board so it overlaps between cam 1 and 2 instead.

frame003-camera001.png
frame003-camera002.png

frame004-camera001.png
frame004-camera002.png

frame005-camera001.png
frame005-camera002.png
```

Would mrcal be able to work out all extrinsics between cameras?

dkogan, 2025-04-23:
This is fine. Have you tried it, and observed issues?

zumpchke, 2025-05-13:
This seemed to work initially. 

When I added the 4th camera, the error becomes pretty large - ~0.8 rms and seemed to keep going up. It appears almost as if the errors were compounding! 

I'm just wondering if there is global optimization applied in the above scenario?

One thing I didn't try is to ensure cam6 and cam0 have some board poses; do you think this is crucial?

dkogan, 2025-05-13:
> I'm just wondering if there is global optimization applied in the
> above scenario?

Yes. There's one big optimization.

> One thing I didn't try is to ensure cam6 and cam0 have some board
> poses; do you think this is crucial?

I don't know enough about your problem. 0.8 pixels rms isn't so big that
I'd think you have some catastrophic problem. More likely it's the usual
issues (camera sync errors, poorly-fitting lens model, stuff moving,
poorly-fitting chessboard, etc).

As an experiment, see what happens if you run a smaller solve with a
subset of your cameras. This might help you chase down the source of
your problem.

zumpchke, 2025-05-13:
Thanks for the feedback! Will do some digging.

Do you have any experience with using a double sided calibration board which could avoid the pairwise calibration technique?

zumpchke, 2025-05-13:
Can you see anything obviously wrong with this calibration? (besides the fact I need way more boards.. :) ) 

[7cam_calib.zip](https://github.com/user-attachments/files/20194120/7cam_calib.zip)

zumpchke, 2025-05-13:
Thought I'd add the corners and invocation to check whether the invocation is such that the bundle adjustment optimization is being done globally

```
mrcal-calibrate-cameras --lensmodel LENSMODEL_OPENCV8 --object-spacing 0.08 --object-width-n 6 --object-height-n 4 --corners-cache corners.vnl --focal 1900 --imagersize 1920 1080 --outdir . frame*-cam000.png frame*-cam001.png frame*-cam002.png frame*-cam003.png frame*-cam004.png frame*-cam005.png frame*-cam006.png
```

[7cam_calib_v2.zip](https://github.com/user-attachments/files/20194726/7cam_calib_v2.zip)

dkogan, 2025-05-14:
You should read the mrcal manual about how to validate your solves: https://mrcal.secretsauce.net/how-to-calibrate.html#interpreting-results

And about how to gather good data: https://mrcal.secretsauce.net/how-to-calibrate.html#dancing

You didn't do the validation. Your solves have a very high uncertainty because you have a small chessboard observed from far away. You want a bigger chessboard; the observations need to cover a bigger region of the imager.

zumpchke, 2025-05-14:
I just tried `mrcal-show-projection-uncertainty camera-0.cameramodel` and I get an uncertainty=1 (yellow) everywhere. I expected some lower uncertainty at the points where I had the board though.

Ah yes, the board is probably too far away because I'm trying to get 2 cameras to see them. Would solving the intrinsics for each camera first then using those models as a seed for the extrinsics solve be helpful? I see similar guidance in your documentation

```
If calibrating multiple cameras, it is impossible to place a calibration object at a location where it's seen by all the cameras and where it's a close-up for all the cameras. So you should get close-ups for each camera individually, and also get observations common to multiple cameras, that aren't necessarily close-ups. The former will serve to define your camera intrinsics, and the latter will serve to define your extrinsics (geometry). Get just far-enough out to create the joint views. If usable joint views are missing, the extrinsics will be undefined, and the solver will complain about a "not positive definite" (singular in this case) Hessian.
```

dkogan, 2025-05-15:
zumpchke ***@***.***> writes:

> I just tried mrcal-show-projection-uncertainty camera-0.cameramodel
> and I get an uncertainty=1 (yellow) everywhere

Yes. Pass `--cbmax 20` to raise the ceiling for color-coding. This solve
is quite bad.

> I expected some lower uncertainty at the points where I had the board
> though

The small-looking board observations aren't enough. Bigger board and/or
closer.

> Would solving the intrinsics for each camera first then using those
> models as a seed for the extrinsics solve be helpful?

No. That creates more problems than it solves.

> I see similar
> guidance in your documentation
>
> ...

Your dataset should have some closeups (for intrinsics; without worrying
about overlap) and some joint observations from further out (for
extrinsics). You already have the latter, but you also need the former.
Run the calibration off that one big dataset together. One big solve.
mrcal will figure out what to do.

zumpchke, 2025-05-15:
Thanks for your generous responses!

> No. That creates more problems than it solves.

Would calculating intrinsics ahead of time cause problems to not be feasible? We need the extrinsics in a space we have a limited time to setup (think DIY mocap in a space we're borrowing), so that makes grabbing enough board poses for a good joint intrinsics and extrinsics difficult. I thought really nailing the intrinsics ahead of time would be a decent tradeoff.

dkogan, 2025-05-15:
If you really need to, you can certainly do that. I have applications where such a 2-step solve is required as well. But then ideally you want the second solve (extrinsics-only) to ingest the uncertainty information from the first solve to intelligently trust some observations more than others. And you want the second solve to report an uncertainty in its output that reflects the uncertainties of both solves. This is something that mrcal cannot do yet, so you lose some of the validation output when you do this.

zumpchke, 2025-05-20:
i will try it out!

Related question - If I pass in `--seed`, and if I am not issuing `--skip-intrinsics-solve` (or `--skip-extrinsics-solve`), does it use the prior data in the optimization?

dkogan, 2025-05-21:
Yes. The `--seed` is where the optimization starts its search. The other options are respected as usual.