dkogan/mrcal #40: Understanding valid intrinsic region

tholzmann, 2025-05-21:
Hello,

I am calibrating a camera where I could reach a quite good RMS error, the projection uncertainty is quite low and also the measurements are distributed quite well over the whole image plane. When undistorting the image, the distortion is also removed well (visually inspected). However, the computed valid intrinsic region is quite small.

My question is now: Should I trust in the uncertainty values, or in the valid intrinsic region? Could you provide an explanation of how the valid intrinsic region is computed exactly and when to trust in it and when not?

See below some analysis output:

RMS reprojection error: 0.1 pixels
Worst residual (by measurement): 0.4 pixels
Noutliers: 758 out of 35200 total points: 2.2% of the data

Projection uncertainty including valid intrinsics region:

[image]

There seem to be some outlier observations, but overall well distributed observations I think:

[image]

Output of mrcal-show-residuals --regional camera-0.cameramodel

[image]

[image]

[image]

Thanks!
Thomas

dkogan, 2025-05-21:
Hello.

The valid-intrinsics region is a somewhat crude measure to evaluate
calibration quality. It exist primarily because

- the full uncertainty logic took a long time to figure out and
  understand and implement, and this valid-intrinsics-region logic came
  first

- the reported uncertainty values are very over-optimistic with lean
  models, and I wanted to have SOME way to quickly evaluate calibration
  results even when using lean models

So the logic is a bit crude, and if you're using a splined model, and
you convinced yourself that the results are good by looking at the other
metrics, then the results are probably good.

The logic is documented in mrcal-calibrate-cameras --help:

  https://github.com/dkogan/mrcal/blob/0d5426b5851be80dd8e51470a0784a73565a3006/mrcal-calibrate-cameras#L204

and implemented here:

  https://github.com/dkogan/mrcal/blob/0d5426b5851be80dd8e51470a0784a73565a3006/mrcal/calibration.py#L1528

tholzmann, 2025-05-22:
Thanks for the additional information. I'm using a lean model, therefore also all the threshold checks in compute_valid_intrinsics_region are executed.

Therefore, one more question:
I don't understand why you are using the std dev of the residuals to weight the thresholds: https://github.com/dkogan/mrcal/blob/0d5426b5851be80dd8e51470a0784a73565a3006/mrcal-calibrate-cameras#L785

Can you comment on this?

Thanks!

tholzmann, 2025-05-22:
FYI I now also tried a splined model, and the projection uncertainties and the valid_intrinsic region is quite similar:

[image]

dkogan, 2025-05-24:
Thomas Holzmann ***@***.***> writes:

> I don't understand why you are using the std dev of the residuals to
> weight the thresholds

Sure. Undern the usual assumptions, the distribution of the residuals
matches the distribution of the input noise.

The scaling allows those parameters in
--valid-intrinsics-region-parameters to be given as unitless numbers, so
the threshold will be more accepting in very noisy problems. The
mean-abs-residuals threshold is NOT scaled to allow rejecting
particularly noisy solutions.

Like I said, this is crude.

dkogan, 2025-05-24:
> FYI I now also tried a splined model, and the projection uncertainties and the valid_intrinsic region is quite similar

Yeah, you're seeing that because your data has VERY good imager coverage. And you do see the uncertainty become a bit worse across the board, and much worse in the extreme corners. That's what you would expect.

closed 2025-05-28 by tholzmann