dkogan/mrcal #42: CAHVORE with E=[0,0,0] vs CAHVOR

SimoneNascivera, 2025-06-27:
Hi,
I noticed a strange behavior of mrcal when dealing with CAHVORE and CAHVOR camera models. 
I have a .cahvore file where "E =      0.000000      0.000000      0.". I am able to project images to a pinhole configuration using image_transformation_map and straight lines are straight.
If I take the .cahvore file and I remove the E line and rename it to .cahvor, and redo the all process, no projection is computed.

Cahvore example:
`
Model = CAHVORE2 = fish-eye

Dimensions = 640 480

C =      0.000000      0.000000      0.000000
A =      0.000000      0.000000      1.000000
H =    274.210101      0.000000    325.2145
V =      0.000000    271.613244    248.234190
O =     -0.005853      0.003144      0.987977
R =      0.000000      0.001016     -0.000278
E =      0.000000      0.000000      0.000000
`
Cahvor  example:
`
Model = CAHVOR2 = fish-eye

Dimensions = 640 480

C =      0.000000      0.000000      0.000000
A =      0.000000      0.000000      1.000000
H =    274.210101      0.000000    325.2145
V =      0.000000    271.613244    248.234190
O =     -0.005853      0.003144      0.987977
R =      0.000000      0.001016     -0.000278
`
Resulting images:
[image]
[image]

Code to undistort:
`import mrcal
import cv2
import os

model = mrcal.cameramodel("genny.cameramodel")
model_pinhole = mrcal.pinhole_model_for_reprojection(model, fit = "corners")
mapxy = mrcal.image_transformation_map(model, model_pinhole, intrinsics_only = True)

base_path = "./img/"
for image_path in os.listdir(base_path): #tqdm.tqdm(os.listdir(base_path)):
    img = cv2.imread(os.path.join(base_path, image_path))
    undistorted = mrcal.transform_image(img, mapxy)
    cv2.imwrite(os.path.join(base_path.replace("img", "img_rect"), image_path), undistorted)`

dkogan, 2025-06-28:
> If I take the .cahvore file and I remove the E line and rename it to
> .cahvor, and redo the all process, no projection is computed.

I don't know what "no projection is computed" means, but removing E=0
doesn't create an equivalent CAHVOR model in all cases. Since that whole
world is a huge mess, the only documentation on this topic is in mrcal:

  https://mrcal.secretsauce.net/docs-2.4/lensmodels.html#org3a3188c

It says

  A LENSMODEL_CAHVORE projection with E = 0 and linearity = 1 is
  equivalent to a LENSMODEL_CAHVOR projection

You have the "fish-eye" flavor of cahvore (first line of your model
file), which has linearity = 0 (see the same mrcal docs, in the previous
paragraph ), so removing the E does NOT give you an equivalent CAHVOR
model. I can't say much about the behavior you're seeing with the CAHVOR
you made, but it should NOT be the same.

And the obligatory: are you SURE you need cahvor(e)? Every single
project would benefit immensely from dropping that immediately,
including whatever you're working on.

And also: are you SURE you want to be converting things to a pinhole
projection? Your straight lines stay straight yes, but your shapes and
sizes become extremely distorted (see the image in this bug report). If
you're doing dense stereo, use the LATLON model instead:

  https://mrcal.secretsauce.net/stereo.html

If you're doing feature tracking or something, doing that in the input
image will give you better estimates (less noise due to remapping and
the noise you do have will homoscedactic, giving you maximum-likelihood
fits).

SimoneNascivera, 2025-07-03:
Forgive me for my ignorance. I was incorrectly assuming what you mentioned, i.e. was not aware that "removing E=0
doesn't create an equivalent CAHVOR model in all cases".

Thank you for the explanation